rusefi/stim #2: Slow Analog Stim module

rusefillc, 2021-07-10:
We need a module to stimulate up to XXX 0-5v analog channels. ECU could have 1K or more pull-up/down.

rusefillc, 2021-07-10:
Would an SPI DAC be the way to go? Does JLC have any? Would SPI DAC produce current needed to accommodate for instance 1K pull-down?

stefanst, 2021-07-10:
MCP4726 DAC is available on JLC.
But addressing become a problem.
Maybe just use one DAC and analog switches?

mck1117, 2021-07-10:
BH2226F-E2 looks great but isn't JLC, plus it's only $2 each, and easy to solder by hand.  Doing a few of those wouldn't be bad at all.

stefanst, 2021-07-10:
No stock of BH2226F-E2 at Mouser or Digikey
BH2221FV-E2 with 12 channels seems available though.

andreika-git, 2021-07-10:
We don't need all analog channels at the same time. We could do the same trick with a multichannel analog demux (see #6), opamp and only 1 DAC output (nucleo already has it).

rusefillc, 2023-07-27:
only static analog pins for now

closed 2023-07-27 by rusefillc