rusefi/stim #3: Hall sensor / digital switch stim

rusefillc, 2021-07-10:
we need up to XXX channels of Hall / switch stimulation

stefanst, 2021-07-10:
Hall should be easy- just an NPN transistor.

rusefillc, 2021-07-10:
but we need 16 of these so some muxing?

stefanst, 2021-07-10:
If memory serves right, more like de-muxing. But yes.

mck1117, 2021-07-10:
or not - just use a 208 pin mcu

rusefillc, 2021-07-10:
208 pin MCU does not exist in 2021.
only 144 pin MCU exist unless you give me available part numbers for Nucleo 176 or else :)

rusefillc, 2023-07-23:
firmware side https://github.com/rusefi/rusefi-hardware/issues/222

next step: selecting eight pins on 144 pin STM32 Nucleo-F429ZI which we can toggle without any interference from other hardware on STM32 Nucleo-F429ZI and not yet used by existing https://github.com/rusefi/rusefi-hardware/tree/main/digital-inputs

rusefillc, 2023-07-26:
CN1-2 would be most convenient
CN1-1 second most convenient

но можно любые

[image]

andreika-git, 2023-07-26:
Let's use these 8 pins on the CN12 connector:

[image]

PE0, PE7-13.

rusefillc, 2023-07-26:
@Qwerty-OFF сделай пожалуйста схематику одного вывода для тестирования цифрового датчика на пине PE0?

Qwerty-OFF, 2023-07-26:
Сделал
[image]

Nadezhda-n, 2023-07-27:
[image]
OUT_HALO_* на  любые ноги 6437288-3?

rusefillc, 2023-07-27:
@Nadezhda-n https://rusefi.com/docs/pinouts/stim/?connector=main&pin=15B и так далее, что digital out - не только hall

Nadezhda-n, 2023-07-27:
> @Nadezhda-n https://rusefi.com/docs/pinouts/stim/?connector=main&pin=15B и так далее, что digital out - не только hall

что

Nadezhda-n, 2023-07-27:
Есть 4 out на NUCLEO-F429ZI, а также 8 hallo
А на разъеме 8 digital out

Nadezhda-n, 2023-07-28:
Теперь я понимаю, что digital out как раз для этих hallo

closed 2023-08-02 by Qwerty-OFF