rusefi/stim #5: Basics

stefanst, 2021-07-10:
Before we start designing circuits, how about a list of requirements?

rusefillc, 2021-07-10:
https://github.com/rusefi/rusefi/wiki/dev-hardware-quality-control#requirements

rusefillc, 2021-07-10:
An interesting thing to approximate would be rate of testing

As is I have no idea what's the time to SEND+ACK a text command via our protocol, and it could be that we would be sending commands to two different devices etc etc.

maybe can between ECU and Stim is really important, that case PC sends Stim one command and Stim exchanges data with ECU directly

stefanst, 2021-07-10:
I don't see rate of test as a real issue 
Let's say we have a max. of 120 i/o which seems reasonable. Even if each test takes one full second, it would only be 2 minutes total. Setup time with flashing firmware twice (before and after testing) will take longer than that. So the time bottleneck would be setup time.

stefanst, 2021-07-13:
Most basic configuration:
- 16x digital potentiometer (different values might be helpful)
- 16 digital out for emulating
  - Cam/ Crank sensors
  - Frequency-based sensors
  - switches
- One fake VR output
- 32 digital in for monitoring digital or PWM outputs
- Maybe 8 PWM -> voltage converters (RC) for simple monitoring of PWM outputs?

closed 2023-07-23 by rusefillc