rusefi/stim #14: 60 pin superseal 1.0 connector

rusefillc, 2023-04-19:
let's use 60 pin superseal 1.0 connector same as https://github.com/rusefi/alphax-8chan/

Power, CAN and all inputs should be moved to that connectors, current wire provisions are not needed

Nadezhda-n, 2023-05-02:
What are the inputs here? :)

Nadezhda-n, 2023-05-02:
> Sorry outputs @Nadezhda-n

[image]
?

Nadezhda-n, 2023-05-02:
Плату нужно увеличивать, верно?
[image]

rusefillc, 2023-07-22:
@rusefillc не совсем - ведь на эти 60 пинов уйдут примерно 36 текущих отверстий справа

48 каналов мне по факту просто не нужно - так что про каналы 33-48 я подумаю чуть позже

а вот 1-32 идут на разъём. Наверное мне стоит сделать картинку желаемой распиновки, и всем станет лучше!

rusefillc, 2023-07-26:
т.е. правый-верхний угол платы заменяется вот на такую распиновку

https://rusefi.com/docs/pinouts/stim/

https://github.com/rusefi/stim/blob/main/digital_inputs/connectors/60pins.yaml

Nadezhda-n, 2023-07-27:
[image]

Nadezhda-n, 2023-07-27:
Куда подключать out4-out7??
[image]

Nadezhda-n, 2023-07-31:
[image]
Такая будет компоновка? 
@rusefillc

rusefillc, 2023-07-31:
если 60 пин развернуть на 180 градусов - то да.

rusefillc, 2023-07-31:
> Куда подключать out4-out7??

 out4-out7 это внутренние трассы, идущие на чипы - их не нужно трогать и выводить на разъёмы

Nadezhda-n, 2023-08-01:
Я плату добавила, вроде с меня на данном этапе все.
@rusefillc @Qwerty-OFF

Qwerty-OFF, 2023-08-02:
Подразумевалось что-то типа OUT_HALL0, OUT_HALL1, сейчас названия выглядят немного странно
[image]

@rusefillc для CAN 120 Ом и защитные диоды нужны?

@rusefillc хочешь ли ты подписи функционала у разъемов с обратной стороны платы или это особый вид наказания специально для меня?

Больше возражений не имею

rusefillc, 2023-08-02:
@Qwerty-OFF спасибо огромное! два тикета сделал

closed 2023-08-08 by Nadezhda-n