rusefi/stim #16: voltage output for static test of analog inputs

rusefillc, 2023-07-22:
stim board should provide 10 points of reference voltage outputs

0.5v
0.9v
0.12v

etc

rusefillc, 2023-07-22:
@Qwerty-OFF please provide resistor divider values for us to divide 5v

Qwerty-OFF, 2023-07-24:
|   | Voltage  | R up |  R down |
| - | ----- | ---- | ----- |
| 1 | 0.5v  | 910 | 100 |
| 2 | 0.9v  | 470 | 100 |
| 3 | 1.3v  | 910 | 330 |
| 4 | 1.7v  | 910 | 470 |
| 5 | 2.1v  | 470 | 330 |
| 6 | 2.5v  | 470 | 470 |
| 7 | 2.9v  | 330 | 470 |
| 8 | 3.3v  | 470 | 910 |
| 9 | 3.7v  | 330 | 910 |
| 10 | 4.1v  | 100 | 470 |

Nadezhda-n, 2023-07-27:
[image]

Nadezhda-n, 2023-07-27:
Extended Part
910
[image]

rusefillc, 2023-07-27:
@Nadezhda-n любой номинал от 890 до 940

жадность проснулась и просит всё-таки 16 каналов - значит любые 6 пар номиналов будут повторяться два раза https://github.com/rusefi/stim/commit/6b318c73dd3a7ee15cf343e752e1aa83c5f0f26b

Nadezhda-n, 2023-07-28:
> @Nadezhda-n любой номинал от 890 до 940
> 
[image]

Nadezhda-n, 2023-07-28:
> жадность проснулась и просит всё-таки 16 каналов - значит любые 6 пар номиналов будут повторяться два раза [6b318c7](https://github.com/rusefi/stim/commit/6b318c73dd3a7ee15cf343e752e1aa83c5f0f26b)

С этим набором может быть 13 разных значений, тебе действительно нужны 16?

Nadezhda-n, 2023-07-28:
> @Nadezhda-n любой номинал от 890 до 940
> 
> жадность проснулась и просит всё-таки 16 каналов - значит любые 6 пар номиналов будут повторяться два раза [6b318c7](https://github.com/rusefi/stim/commit/6b318c73dd3a7ee15cf343e752e1aa83c5f0f26b)

И как расставлять 5В?
[image]

Nadezhda-n, 2023-08-01:
REF14-16 удаляем?

rusefillc, 2023-08-01:
> REF14-16 удаляем?

не понимаю, почему удаляем? они же есть на распиновке? https://rusefi.com/docs/pinouts/stim/?pin=30C

rusefillc, 2023-08-01:
> REF14-16 удаляем?

примерно любые пары резисторов из набора Володиного

Nadezhda-n, 2023-08-01:
> И как расставлять 5В? 

по 4 на каждые у обоих разъемов или на один разъем 5V_1, а на другой 5V_2?

Qwerty-OFF, 2023-08-02:
[image]
В группах вероятно неразличимы между собой будут REF3 и REF4, REF7 и REF8, если это важно

rusefillc, 2023-08-02:
> [image] В группах вероятно неразличимы между собой будут REF3 и REF4, REF7 и REF8, если это важно

ref4 насколько тревиально поднять до 1.4v? ещё есть вариант резисторы поставить 1% - эта плата достаточно важна для этого

@Qwerty-OFF давай поставим резисторы 1% и попробуем вообще улучшить, если малыми усилиями

Qwerty-OFF, 2023-08-02:
@Nadezhda-n ты сможешь выдумать 16 пар резисторов 1% любого номинала какой найдешь на JLC? Так чтобы:
- все соседние напряжение отличались хотя бы на 10%
- сумма резисторов была 500-1500 Ом

Можно ориентироваться на такой ряд напряжений
4.25
3.61
3.07
2.61
2.22
1.89
1.60
1.36
1.16
0.98
0.84
0.71
0.60
0.51
0.44
0.37

rusefillc, 2023-08-02:
> @Nadezhda-n ты сможешь выдумать 16 пар резисторов 1% любого номинала 

если за 30 минут не найдётся, я бы просто бы нашёл 1% резисторы как они есть. вроде у нас АЦП всё-таки 10 бит и модные опорные 5 вольт?

Nadezhda-n, 2023-08-02:
> @Nadezhda-n ты сможешь выдумать 16 пар резисторов 1% любого номинала какой найдешь на JLC? Так чтобы:
>
> * все соседние напряжение отличались хотя бы на 10%
> * сумма резисторов была 500-1500 Ом
>
> Можно [...]

[image]
Вот что получилось

Nadezhda-n, 2023-08-02:
линейка резисторов в омах
[image]

Qwerty-OFF, 2023-08-02:
В 10% кажется в нескольких местах не попадает (если посчитать напряжения с десятыми), но 5% есть. Думаю можно так оставить.

closed 2023-08-08 by Nadezhda-n