rusefi/stim #18: let's have china solder CN1-2

rusefillc, 2023-07-27:


rusefillc, 2023-07-31:
если у JLC нет идеальных разъёмов - давайте обсудим, насколько будет просто запаять хотя бы частичный разъём - может быть скобинировать из нескольких разъёмов с пропусками

сейчас добавлю картинку, чтоб повысить шанс меня понять

rusefillc, 2023-07-31:
[image]

rusefillc, 2023-08-01:
> Я не знаю, кто такие "x by 2" headers

я про разъёмы на U11, там 6 штук разного размера например 8x2 и 10x2 и ещё какие-то более длинные

Nadezhda-n, 2023-08-01:
> > Я не знаю, кто такие "x by 2" headers
> 
> я про разъёмы на U11, там 6 штук разного размера например 8x2 и 10x2 и ещё какие-то более длинные

Вместо U11 стоят разъемы 2,54 стандартные и нужно подобрать перечень с jls, чтобы поставиьь их во все отверстия футпринта?

rusefillc, 2023-08-01:
@Nadezhda-n на 90% да!

если всё легко и у них есть все разъёмы - то просто ставим

нюансы
* если какой-то разъём у нас не испольщуется вообще - аргумент не ставить, чтоб проще было вставлять-вынимать U11
* если правильных огромных у JLC нет, то большую серцию есть идея разделить логически и попросить JLC напаять хотя бы самые важные для нас сектора

Nadezhda-n, 2023-08-01:
> @Nadezhda-n на 90% да!
>
> если всё легко и у них есть все разъёмы - то просто ставим
>
> нюансы
>
> * если какой-то разъём у нас не испольщуется вообще - аргумент не ставить, чтоб проще было [...]

[image]

В первой линии у нас используется 5 пинов, то есть можно взять PLD-10 или нужно побольше? (Кстати, вилки нужны?)

Nadezhda-n, 2023-08-01:
[image]
C92264 DS1023-2*17SF11
C92264 DS1023-2*17SF11

C92267 DS1023-2*8SF11
C132129 DS1023-2*15SF11

C92266 DS1023-2*10SF11
C92264 DS1023-2*17SF11

C92266 DS1023-2*10SF11
C92266 DS1023-2*10SF11
C132127 DS1023-2*13SF11

Qwerty-OFF, 2023-08-02:
[image]

Nadezhda-n, 2023-08-02:
> [image]

[image]

Суть в том, что корпус разъема немного больше (0,5+-0,3мм), и вот вплотную три подряд разъема могут не влезть, поэтому может все таки сделаем с зазорами, как я кидала выше?

Qwerty-OFF, 2023-08-02:
Соседние разъемы не будут паяться, я в BOM поставил только те, что нужны

closed 2023-08-08 by Nadezhda-n