rusefi/stim #24: manually fix units (at least in drill files)

rusefillc, 2023-08-06:
Automations is failing because of mm<>inches wrong setting

See https://github.com/andreika-git/hellen-one/issues/320

rusefillc, 2023-08-06:
@Nadezhda-n looks like inches bad https://github.com/andreika-git/gerbmerge/blob/43eb62470565f5f408cf8094047624bc43f90183/gerbmerge/jobs.py#L388-L390 mm good?

@Nadezhda-n please change units to mm manually so that we could move forward

Nadezhda-n, 2023-08-08:
> @Nadezhda-n looks like inches bad https://github.com/andreika-git/gerbmerge/blob/43eb62470565f5f408cf8094047624bc43f90183/gerbmerge/jobs.py#L388-L390 mm good?
> 
> @Nadezhda-n please change units to mm manually so that we could move forward

Я не понимаю, где я должна изменить дюймы в мм, я ведь не делаю выгрузку вручную? А работаю я в мм

rusefillc, 2023-08-08:
я вот что осознал: в других местах выгрузка работает
похоже нужно один раз сделать выгрузку drill руками, и это просто поменяет настройки в проекте
и автоматические выгрузки тоже начнут работать, если я угадал проблему

Qwerty-OFF, 2023-08-08:
Кажется дело было в старых герберах в той же папке, использовались они вместо свежих

rusefillc, 2023-08-08:
https://github.com/andreika-git/hellen-one/issues/321 created to clean gerbers folder prior to export