rusefi/stim #30: CAN and USB share same pins

rusefillc, 2023-08-20:
печаль
```
#define EFI_USB_SERIAL_PORT GPIOA
#define EFI_USB_SERIAL_PIN_DM 11
#define EFI_USB_SERIAL_PIN_DP 12
```

```
#define CAN_PORT GPIOA
#define CAN_PIN_RX 11
#define CAN_PIN_TX 12
```

rusefillc, 2023-08-20:
попробую перерезать CAN_TX CAN_RX и кину сопли на

[image]

rusefillc, 2023-08-21:
на схеме CN11 снизу обрезаны два GND пина, для уточнения вот эти пины используем для CAN шины теперь

[image]

rusefillc, 2023-08-21:
там возможно эти пины два раза выведены - ты смотришь на CN9

я как-то так получилось использовал другие на CN11, давай уже для простоты используем их

https://os.mbed.com/platforms/ST-Nucleo-F429ZI/

[image]

[image]

Nadezhda-n, 2023-08-21:
> попробую перерезать CAN_TX CAN_RX и кину сопли на
> 
> [image]

Резать еще что-то надо? 
Или дублирует выводы просто?

rusefillc, 2023-08-21:
на плате 0.2 я отрезал старые CAN трассы

на rev C трассы будут вести только на PD0/PD1 и не будет CAN трасс на PA10/PA11

Nadezhda-n, 2023-08-21:
> на CN11, давай уже для прост

оки
[image]

Qwerty-OFF, 2023-12-10:
[image]

closed 2023-12-10 by Qwerty-OFF