rusefi/stim #35: second CAN bus

rusefillc, 2023-08-22:


Qwerty-OFF, 2023-12-10:
PB6/PB12. OUT0 перемещен с PB12 на PD8.
Установить чип CAN и вывести CAN на via?

[image]

rusefillc, 2023-12-10:
> Установить чип CAN 

да

> вывести CAN на via?

пока да, у нас там есть via на разъёме - но мы пока не будет на разъём

Qwerty-OFF, 2023-12-10:
[image]

rusefillc, 2024-01-04:
>  OUT0 перемещен с PB12 на PD8.

жутко извиняюсь, но хочется уточнить: это единственное изменение распиновки и несовместимость между ревизиями A и С?

Qwerty-OFF, 2024-01-04:
Кажется больше ничего не трогали. Есть вероятность путаницы номеров CAN1/CAN2, если ты пытаешься запустить оба сразу

closed 2024-10-18 by rusefillc