rusefi/stim #45: rev C вся нуклео не включается от 12 вольт

rusefillc, 2024-10-18:
на rev C есть кнопка

при питании от 12 вольт в одном положении ничего не светится

при питании от 12 вольт в другом положении мигает LD4 наверху нуклео, но на основном блоке нуклео ничего не светится

прошивка stim начинает мигать голубым только при подключении microUSB

Qwerty-OFF, 2024-10-18:
[image]
Джампер сюда. Питание 12В максимум

Qwerty-OFF, 2024-10-19:
Мы будем использовать E5V
[image]

Qwerty-OFF, 2024-10-19:
Поставил стабилизатор на 5В

rusefillc, 2024-11-01:
extra nucleo pin for 5v

[image]

rusefillc, 2024-11-01:
JP3 подвинул, как на картинке

третий пин J20 припаял, как на картинке

LD4 мигает
LD6 не горит, прошивка не стартует @Qwerty-OFF

Qwerty-OFF, 2024-11-02:
JP3 передвинь в E5V, та картинка была для внешних 12В

closed 2024-11-02 by rusefillc