rusefi/stim #48: just damaged rev C number 2 while playing with 10R 20W between 19C and 20C

rusefillc, 2024-10-26:
we were trying to add real load to improve detection of bad ECUs

specific board was passing good with 0.58v MAP as expected using 8B ``Reference voltage out 2 (MAP) 0.6v``

suddenly voltage check started to fail

```
 * BAD channel 4 (MAP): voltage 0.363921568 (raw 32) not in range (0.670109504..0.000000000)
 ***********************************************
 ***********************************************
 * BAD channel 4 (MAP): voltage 0.363921568 (raw 32) not in range (0.670109504..0.000000000)
 ***********************************************
 ***********************************************
 * BAD channel 4 (MAP): voltage 0.363921568 (raw 32) not in range (0.670109504..0.000000000)
```

[image]

jharvey, 2024-10-26:
@rusefillc @Qwerty-OFF 
While TC4427 can drive around 1.5A, the MUX CD4067BM96 is only rated for around +/- 20mA. With a recommended max of +/- 10ma continuous. It appears the MUX is not protected against overcurrent, so it's likely the 10R damaged that channel of the MUX, perhaps other channels as well.

rusefillc, 2024-11-02:
my bad that was a software issue

closed 2024-11-02 by rusefillc